AversivePlusPlus/AversivePlusPlus #10: Error while compiling Sasiae module

arodrigue003, 2016-02-15:
I got this error when I try to compile sasiae module on my computer : 
`../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::sendData(char const*) »:
client_thread.cpp:(.text+0xb2): référence indéfinie vers « QMutex::lock() »
client_thread.cpp:(.text+0xe6): référence indéfinie vers « QMutex::unlock() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::sync() »:
client_thread.cpp:(.text+0x100): référence indéfinie vers « QMutex::lock() »
client_thread.cpp:(.text+0x115): référence indéfinie vers « QSemaphore::available() const »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::ClientThread() »:
client_thread.cpp:(.text+0x3d4): référence indéfinie vers « QThread::QThread(QObject*) »
client_thread.cpp:(.text+0x3ec): référence indéfinie vers « QMutex::QMutex(QMutex::RecursionMode) »
client_thread.cpp:(.text+0x3fa): référence indéfinie vers « QSemaphore::QSemaphore(int) »
client_thread.cpp:(.text+0x408): référence indéfinie vers « QMutex::QMutex(QMutex::RecursionMode) »
client_thread.cpp:(.text+0x40f): référence indéfinie vers « QMapDataBase::shared_null »
client_thread.cpp:(.text+0x421): référence indéfinie vers « QMutex::QMutex(QMutex::RecursionMode) »
client_thread.cpp:(.text+0x4c5): référence indéfinie vers « QMutex::~QMutex() »
client_thread.cpp:(.text+0x4de): référence indéfinie vers « QMutex::~QMutex() »
client_thread.cpp:(.text+0x4ee): référence indéfinie vers « QSemaphore::~QSemaphore() »
client_thread.cpp:(.text+0x4fe): référence indéfinie vers « QMutex::~QMutex() »
client_thread.cpp:(.text+0x50b): référence indéfinie vers « QThread::~QThread() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::~ClientThread() »:
client_thread.cpp:(.text+0x562): référence indéfinie vers « QMutex::~QMutex() »
client_thread.cpp:(.text+0x574): référence indéfinie vers « QMutex::~QMutex() »
client_thread.cpp:(.text+0x57d): référence indéfinie vers « QSemaphore::~QSemaphore() »
client_thread.cpp:(.text+0x586): référence indéfinie vers « QMutex::~QMutex() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::registerDevice(SASIAE::Device const&, std::function<void (char*)> const&) »:
client_thread.cpp:(.text+0x69c): référence indéfinie vers « QMutex::lock() »
client_thread.cpp:(.text+0x6b7): référence indéfinie vers « QMutex::unlock() »
client_thread.cpp:(.text+0x6d3): référence indéfinie vers « QMutex::unlock() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::run() »:
client_thread.cpp:(.text+0x802): référence indéfinie vers « QSemaphore::release(int) »
client_thread.cpp:(.text+0x86d): référence indéfinie vers « QSemaphore::release(int) »
client_thread.cpp:(.text+0x8f4): référence indéfinie vers « QMutex::lock() »
client_thread.cpp:(.text+0x925): référence indéfinie vers « QMutex::unlock() »
client_thread.cpp:(.text+0x9c4): référence indéfinie vers « QMutex::unlock() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::sync() »:
client_thread.cpp:(.text+0x124): référence indéfinie vers « QMutex::unlock() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « SASIAE::ClientThread::~ClientThread() »:
client_thread.cpp:(.text+0x58f): référence indéfinie vers « QThread::~QThread() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QString::QString(char const*) »:
client_thread.cpp:(.text._ZN7QStringC2EPKc[_ZN7QStringC5EPKc]+0x24): référence indéfinie vers « QString::fromAscii_helper(char const*, int) »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QString::~QString() »:
client_thread.cpp:(.text._ZN7QStringD2Ev[_ZN7QStringD5Ev]+0x1f): référence indéfinie vers « QArrayData::deallocate(QArrayData*, unsigned long, unsigned long) »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QMapData<QString, std::function<void (char*)> >::findNode(QString const&) const »:
client_thread.cpp:(.text._ZNK8QMapDataI7QStringSt8functionIFvPcEEE8findNodeERKS0_[_ZNK8QMapDataI7QStringSt8functionIFvPcEEE8findNodeERKS0_]+0x1c): référence indéfinie vers « operator<(QString const&, QString const&) »
client_thread.cpp:(.text._ZNK8QMapDataI7QStringSt8functionIFvPcEEE8findNodeERKS0_[_ZNK8QMapDataI7QStringSt8functionIFvPcEEE8findNodeERKS0_]+0x43): référence indéfinie vers « operator<(QString const&, QString const&) »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QMapData<QString, std::function<void (char*)> >::destroy() »:
client_thread.cpp:(.text._ZN8QMapDataI7QStringSt8functionIFvPcEEE7destroyEv[_ZN8QMapDataI7QStringSt8functionIFvPcEEE7destroyEv]+0x1f): référence indéfinie vers « QMapDataBase::freeTree(QMapNodeBase*, int) »
client_thread.cpp:(.text._ZN8QMapDataI7QStringSt8functionIFvPcEEE7destroyEv[_ZN8QMapDataI7QStringSt8functionIFvPcEEE7destroyEv]+0x28): référence indéfinie vers « QMapDataBase::freeData(QMapDataBase*) »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QMapData<QString, std::function<void (char*)> >::createNode(QString const&, std::function<void (char*)> const&, QMapNode<QString, std::function<void (char*)> >*, bool) »:
client_thread.cpp:(.text._ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb[_ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb]+0x20): référence indéfinie vers « QMapDataBase::createNode(int, int, QMapNodeBase*, bool) »
client_thread.cpp:(.text._ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb[_ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb]+0x50): référence indéfinie vers « qt_assert(char const*, char const*, int) »
client_thread.cpp:(.text._ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb[_ZN8QMapDataI7QStringSt8functionIFvPcEEE10createNodeERKS0_RKS4_P8QMapNodeIS0_S4_Eb]+0xa4): référence indéfinie vers « QMapDataBase::freeNodeAndRebalance(QMapNodeBase*) »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QMap<QString, std::function<void (char*)> >::detach() »:
client_thread.cpp:(.text._ZN4QMapI7QStringSt8functionIFvPcEEE6detachEv[_ZN4QMapI7QStringSt8functionIFvPcEEE6detachEv]+0xf): référence indéfinie vers « QMapDataBase::createData() »
client_thread.cpp:(.text._ZN4QMapI7QStringSt8functionIFvPcEEE6detachEv[_ZN4QMapI7QStringSt8functionIFvPcEEE6detachEv]+0x5c): référence indéfinie vers « QMapDataBase::recalcMostLeftNode() »
../../../libAversive++.a(client_thread.cpp.o): dans la fonction « QMap<QString, std::function<void (char*)> >::insert(QString const&, std::function<void (char*)> const&) »:
client_thread.cpp:(.text._ZN4QMapI7QStringSt8functionIFvPcEEE6insertERKS0_RKS4_[_ZN4QMapI7QStringSt8functionIFvPcEEE6insertERKS0_RKS4_]+0x4b): référence indéfinie vers « operator<(QString const&, QString const&) »
client_thread.cpp:(.text._ZN4QMapI7QStringSt8functionIFvPcEEE6insertERKS0_RKS4_[_ZN4QMapI7QStringSt8functionIFvPcEEE6insertERKS0_RKS4_]+0x80): référence indéfinie vers « operator<(QString const&, QString const&) »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTIN6SASIAE12ClientThreadE[_ZTIN6SASIAE12ClientThreadE]+0x18): référence indéfinie vers « typeinfo for QThread »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x10): référence indéfinie vers « QThread::metaObject() const »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x18): référence indéfinie vers « QThread::qt_metacast(char const*) »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x20): référence indéfinie vers « QThread::qt_metacall(QMetaObject::Call, int, void**) »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x38): référence indéfinie vers « QThread::event(QEvent*) »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x40): référence indéfinie vers « QObject::eventFilter(QObject*, QEvent*) »
../../../libAversive++.a(client_thread.cpp.o):(.data.rel.ro._ZTVN6SASIAE12ClientThreadE[_ZTVN6SASIAE12ClientThreadE]+0x48): référence indéfinie vers « QObject::timerEvent(QTimerEvent*) »
[... cut ...]
Makefile:137: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:36: recipe for target 'sasiae' failed
make: *** [sasiae] Error 2
`
I succeded to compile the module if I change the file /module/sasiae/test/CMakeLists.txt to : 

> cmake_minimum_required(VERSION 2.8.11)
> find_package(Qt5Core)
> find_package(Qt5Widgets)

> add_executable(sasiae_test test.cpp)
> target_link_libraries(sasiae_test sasiae Qt5::Core Qt5::Widgets Aversive++ -Wl,-whole-archive startup -Wl,-no-whole-archive)

astralien3000, 2016-02-15:
Thanks for your feedback !
You did not specify that, but I know that you are under Ubuntu. Which version ?

The problem is that the library don't need Qt5Widget normally... I will try to get that bug on a docker. But anyway, if I don't, I guess adding Qt5Widget will not create problems.

arodrigue003, 2016-02-15:
I use Ubuntu 15.04 with linux kernel  4.3.0-rc2-wl-ath.
I use the qt5-default package from the official ubuntu store

I tried to just add the `find_package(Qt5Core)` line and by changing `Qt5Core` by `Qt5::Core` in the target_link_libraries but it don't compile and output the same error.
I succeed to compile the sasiae module only if Cmake uses the Qt5::Widgets lib.

astralien3000, 2016-02-15:
I got the bug on a docker. Ubuntu seems to need a very precise syntax for the link with Qt5.
The bug seems to be also present for the last version of Ubuntu (15.10).

astralien3000, 2016-02-15:
Ok, the problem should be solved (I used your solution). Can you confirm ?

arodrigue003, 2016-02-15:
Now it works, thanks

closed 2016-02-16 by astralien3000