AversivePlusPlus/AversivePlusPlus #12: Getting rid of SASIAE::Aversive::sync()

astralien3000, 2016-02-15:
This is the last non portable instruction remaining in the SASIAE interface.

Elioty, 2016-02-15:
The problem here is if we let the user just use:
```C++
while(1) {
  // Do stuff
}
```
As people usually do on microcontroller, it will just burn the CPU.

Though, we could work something out of pthread library, which provides suspend and continue functions (unfortunately, QThread from Qt5 doesn't provide such functionalities). But there still needs a way to let the client thread get the main thread handle to manage its execution (still have an `Aversive::init` like function ? Or a way to run some code before `main`is called?). Moreover, about timed events (for scheduling purpose), I am not sure implementing such solution wouldn't create some race conditions or other multithreading problems. Got to keep an eye on that matter.

astralien3000, 2016-02-16:
The Aversive::init function is not compulsory since it is called when the user creates the first device.

I agree with you that a way to pause the main thread (and for microcontrollers, to go into sleep mode) should be provided. But it is not the purpose of Aversive::sync. It's only purpose is to slow down the execution when simulating. The intent of the user is not to pause the program.

More over, if any algorithm uses loops and is "long" enough, it may cause problems. Because currently, if the robot's program don't call Aversive::sync, the simulator freezes. So, maybe we don't want to slow down the robot's program like this...

Elioty, 2016-02-16:
> The Aversive::init function is not compulsory since it is called when the user creates the first device.

True that. Let's keep it that way. Would be SASIAE::Aversive::init now though, right?

> I agree with you that a way to pause the main thread (and for microcontrollers, to go into sleep mode) should be provided.

Yes, we should at one point. But we will then have to see how to propose a common API on microcontroller and on simulator for that matter, and how to manage the sleep "depth" (ie. IRQ that wake up the CPU or not).

> But it is not the purpose of Aversive::sync. It's only purpose is to slow down the execution when simulating. The intent of the user is not to pause the program.

Yes,  you are right here too.

> More over, if any algorithm uses loops and is "long" enough, it may cause problems. Because currently, if the robot's program don't call Aversive::sync, the simulator freezes. So, maybe we don't want to slow down the robot's program like this...

It's true. Though the microcontrollers we target are way less powerful than pc we run simulation on so if it's freeze the simulator, it is certainly going to freeze the real-life application too (though the microcontroller can still stop and process IRQ handlers). And not having anything to sync the application execution and the simulator will just make the application burn the CPU as said previously, which is not good for your overall framerate too.

So as a solution, we could just let the main thread run as much as it wants/needs, use pthread suspend and continue when the client thread receive event to process like IRQ (to stop the main thread just like on real microcontroller applications), and provide the sleep functions which would make the main thread stop until a new event is received and processed by the client thread (easily doable with condition variables).

Another way would be to remove the client thread entirely and use signals. So instead of having a client thread blocking on a read operation (on stdin), we would just have the main thread, which would, at one point in the init, set up the signal handler. The simulator on the other hand then just push its messages on our application input (just like nowadays) and then send a signal to make the application main thread stop and let the signal handler process the messages. We still need the sleep functions to let the main thread release the CPU, and any blocking method can easily be used for that, as long as it returns when a signal is received (see man 3 errno, error EINTR and man 7 signal, "Interruption of system calls and library functions by signal handlers" paragraph).
However, I am not sure this second solution would be doable for all application/simulator protocols.

closed 2016-09-21 by astralien3000