AversivePlusPlus/AversivePlusPlus #18: A module for SCS15 servos

astralien3000, 2016-03-15:
Here is the [Arduino library](http://www.rcservozone.com/attach/Arduino%20library%20SCServo/SCServoArduinoLib%20for%20SC10%20and%20SM10_150612.zip)

 - It should be modified to be able to use others Serials (and even several serials).
 - Asynchronous mode ?

Elioty, 2016-03-15:
I don't know how their communication protocol works but asynchronous mode could be a great thing! For example, if the protocol specifies the servo does not answer anything after being given an order (like move to that given angle ; or if there is a way to tell the servo not to answer), the microcontroller could just send multiple orders in a row, for the different servos on the same bus, without waiting for any answer. It would be a great thing for low-power use case, if the bus controller (in that case UART controller) has an hardware FIFO, the driver can fill it with the orders and the application/microcontroller can just go to sleep afterwards. And even for queries you expect an answer from, it could be great to let the application do other thing or go into sleep mode. However, it starts to bring some complexity and the possible need of mutual exclusion mechanisms. AversiveOS++ could become a thing in the future.

Anyway, apart from speculations and/or future project, it's great to start getting physical/mechanical device drivers (that are a bit more complex than just a generic PWM-driven motor) in Aversive++ :)

astralien3000, 2016-03-16:
> if the protocol specifies the servo does not answer anything after being given an order (like move to that given angle ; or if there is a way to tell the servo not to answer), the microcontroller [...]

The problem is as far as I know, the protocol does not provide such a functionality. There is no documentation, so it is hard to find information while I haven't worked seriously on it.

> Anyway, apart from speculations and/or future project, it's great to start getting physical/mechanical device drivers (that are a bit more complex than just a generic PWM-driven motor) in Aversive++ :)

Yes ! Finally !

astralien3000, 2016-05-12:
Done, will be merged with the PR #19.

closed 2016-05-12 by astralien3000