AversivePlusPlus/AversivePlusPlus #20: Improving the module system

astralien3000, 2016-06-07:
Currently, the modules are separated, but unable to be compiled as standalone, and the dependency management is not good (you need to add the modules in a particular order to be compiled correctly).

The fact is CMake is not good a doing this. But I have found a package manager that can be integrated to our current build chain, and simplify it a lot : [conan](https://github.com/conan-io/conan).

I think it is worth to at least try to make every modules as a conan package (see the branch dev-conan in each module).

Tasks: 
 - [ ] Make every module a conan package
 - [ ] Make the whole library a conan package
 - [ ] Tests/debugs

Elioty, 2016-06-07:
I heard about conan, in a good way! We can give it a try ;)

astralien3000, 2016-06-07:
Cool !
The problem is that I need to pass the information of target (mcu/native/etc...) across the modules, which I can do (easily) in a "custom way" with the current conan, but I would like to help their project to have a standard way to do so.
That is what I discuss on [this issue](https://github.com/conan-io/conan/issues/86#issuecomment-224337160), if you want to take part.

Elioty, 2016-06-07:
I read that other issue from conan repo, their project is cool AND they listen to feedbacks, how great :)

astralien3000, 2016-06-08:
Here it is !
[All modules have been ported to conan](https://github.com/AversivePlusPlus/AversivePlusPlus/tree/dev-conan).

astralien3000, 2016-10-23:
Done

closed 2016-10-23 by astralien3000